electro-smith/DaisyDuino #6: issues with analogRead in PatchMajorChord example

julianstein, 2020-06-06:
Just received my Daisy -- have the audio examples running fine with arduino, though having issues running audio with analog sensor input. For instance -- in the PatchMajorChord example, the knob input on _A0_ prints out only as 512 and doesn't seem to have an effect on the audio. 

My circuit is fine as I tested it in a separate patch and the analogRead prints as it should without the Daisy library -- Is there a step I am missing in the process? Below is a screenshot of my arduino settings. 

[image]

Thanks,

Julian

stephenhensley, 2020-06-08:
@julianstein 

Can you provide a photo/description of how your circuit is hooked up?

Your settings look like they should be fine.

The first thing that comes to mind is that your AGND and DGND may not be connected.

julianstein, 2020-06-10:
@stephenhensley yes -- issue was with the grounds not being connected, wasn't aware that this was necessary. Thanks ! 

Julian

closed 2020-06-10 by julianstein