electro-smith/DaisyDuino #8: STM32_Programmer_CLI not found

wraybowling, 2020-07-09:
According to the stm32duino wiki, [upload methods that don't require stm32cube to be installed have been deprecated](https://github.com/stm32duino/wiki/wiki/Upload-methods#stm32cubeprogrammer) so I think we should also call out that in order to use this library, you have to install stm32cube.

stephenhensley, 2020-07-09:
We do have the following in the [Getting Started (Arduino Edition)](https://github.com/electro-smith/DaisyWiki/wiki/1a.-Getting-Started-(Arduino-Edition)) section of the wiki. If you were looking somewhere else, we can add notice, and/or a link to the wiki from there as well.

>**Make sure to perform the "extra step" of installing STM32CubeProg, as that application is required for uploading firmware to the Daisy.** You can return to this Wiki page after that step.

wraybowling, 2020-07-09:
Ah-ha. Yeah, i was following the Getting Started section inside the README of this repo. https://github.com/electro-smith/ArduinoAudio#getting-started Definitely should link to the wiki page from there. That'd solve all future confusion 👍

andrewikenberry, 2020-07-09:
I'll get the readme updated today

andrewikenberry, 2020-07-09:
readme is updated and now points to the DaisyWiki getting started page

closed 2020-07-09 by andrewikenberry