electro-smith/DaisyDuino #21: Linker error when using DAISY.end()

pigatron-industries, 2020-10-31:
Hi, when using DAISY.end() to stop the audio callback I get the following error:

```undefined reference to 'dsy_audio_stop'```

I think becasue this is declared as **dsy_audio_stop** in hid_audio.h but is called **audio_stop** in hid_audio.c

An easy workaround for now is to declare it in my own file as follows and call audio_stop directly:

```
extern "C" {
    void audio_stop(uint8_t intext);
}

audio_stop(DSY_AUDIO_INTERNAL);
```

stephenhensley, 2021-01-18:
Good catch! And sorry for the late response! We'll look into this and get it fixed ASAP.

stephenhensley, 2021-02-12:
This is/was resolved in PR #24 

If you are working from git you can simply pull to master. Otherwise we'll have a new release very soon with this fixed up for installations using the Arduino Library Manager.

closed 2021-02-12 by stephenhensley