electro-smith/DaisyDuino #27: SAI code within library breaks compilation with latest stm32 board support update (v2.0)

stephenhensley, 2021-04-22:
Linker error due to redefinition of the HAL files. 

We may be able to simply  remove the HAL SAI files from the utily folder, as those are the files that are triggering the issue.

The errors:

```
"C:\\Users\\shensley\\AppData\\Local\\Temp\\arduino_build_432111\\core\\core.a" -lc -Wl,--end-group -lm -lgcc -lstdc++
c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_Init':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_Init+0x0): multiple definition of `HAL_SAI_Init'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_Init+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_InitProtocol':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_InitProtocol+0x0): multiple definition of `HAL_SAI_InitProtocol'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_InitProtocol+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_DeInit':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_DeInit+0x0): multiple definition of `HAL_SAI_DeInit'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_DeInit+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_Transmit':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_Transmit+0x0): multiple definition of `HAL_SAI_Transmit'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_Transmit+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_Receive':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_Receive+0x0): multiple definition of `HAL_SAI_Receive'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_Receive+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_Transmit_IT':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_Transmit_IT+0x0): multiple definition of `HAL_SAI_Transmit_IT'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_Transmit_IT+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_Receive_IT':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_Receive_IT+0x0): multiple definition of `HAL_SAI_Receive_IT'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_Receive_IT+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_DMAPause':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_DMAPause+0x0): multiple definition of `HAL_SAI_DMAPause'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_DMAPause+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_DMAResume':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_DMAResume+0x0): multiple definition of `HAL_SAI_DMAResume'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_DMAResume+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_DMAStop':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_DMAStop+0x0): multiple definition of `HAL_SAI_DMAStop'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_DMAStop+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_Abort':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_Abort+0x0): multiple definition of `HAL_SAI_Abort'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_Abort+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_Transmit_DMA':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_Transmit_DMA+0x0): multiple definition of `HAL_SAI_Transmit_DMA'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_Transmit_DMA+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_Receive_DMA':

stm32yyxx_hal_sai.c:(.text.HAL_SAI_Receive_DMA+0x0): multiple definition of `HAL_SAI_Receive_DMA'; C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\DaisyDuino\utility\stm32h7xx_hal_sai.c.o:stm32h7xx_hal_sai.c:(.text.HAL_SAI_Receive_DMA+0x0): first defined here

c:/users/shensley/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\shensley\AppData\Local\Temp\arduino_build_432111\libraries\SrcWrapper\HAL\stm32yyxx_hal_sai.c.o: in function `HAL_SAI_EnableTxMuteMode':

[... cut ...]
Using library MIDI_Library at version 5.0.2 in folder: C:\Users\shensley\Documents\Arduino\libraries\MIDI_Library 
Using library DaisyDuino at version 1.3.0 in folder: C:\Users\shensley\Documents\Arduino\libraries\DaisyDuino 
Using library Wire at version 1.0 in folder: C:\Users\shensley\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\Wire 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\shensley\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper 
exit status 1
Error compiling for board Generic STM32H7 Series.
```

tmarasco125, 2021-04-28:
Hi! I just tried this (deleting the stm37xx_hal_sai) libraries from the util folder, and I got a different error. This is after attempting to upload the Encoder example form the Pod collection:
`Arduino: 1.8.13 (Mac OS X), Board: "Generic STM32H7 Series, Daisy Seed, STM32CubeProgrammer (DFU), Enabled (generic 'Serial'), CDC (generic 'Serial' supersede U(S)ART), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

In file included from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/utility/sai.h:7,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/utility/audio.h:4,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/AudioClass.h:4,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/AudioClass.cpp:1:
/Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_sai.h:233:3: error: 'DMA_HandleTypeDef' does not name a type; did you mean '__SAI_HandleTypeDef'?
  233 |   DMA_HandleTypeDef         *hdmatx;      /*!< SAI Tx DMA handle parameters */
      |   ^~~~~~~~~~~~~~~~~
      |   __SAI_HandleTypeDef
/Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_sai.h:235:3: error: 'DMA_HandleTypeDef' does not name a type; did you mean '__SAI_HandleTypeDef'?
  235 |   DMA_HandleTypeDef         *hdmarx;      /*!< SAI Rx DMA handle parameters */
      |   ^~~~~~~~~~~~~~~~~
      |   __SAI_HandleTypeDef
In file included from /Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/board.h:9,
                 from /Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/wiring.h:41,
                 from /Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/Arduino.h:36,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/DaisyDuino.h:4,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/AudioClass.cpp:2:
/Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/backup.h: In function 'void resetBackupDomain()':
/Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/backup.h:74:3: error: '__HAL_RCC_BACKUPRESET_FORCE' was not declared in this scope
   74 |   __HAL_RCC_BACKUPRESET_FORCE();
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/backup.h:75:3: error: '__HAL_RCC_BACKUPRESET_RELEASE' was not declared in this scope
   75 |   __HAL_RCC_BACKUPRESET_RELEASE();
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/board.h:17,
                 from /Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/wiring.h:41,
                 from /Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/Arduino.h:36,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/DaisyDuino.h:4,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/AudioClass.cpp:2:
/Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/uart.h: At global scope:
/Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/uart.h:69:3: error: 'UART_HandleTypeDef' does not name a type; did you mean 'SAI_HandleTypeDef'?
   69 |   UART_HandleTypeDef handle;
      |   ^~~~~~~~~~~~~~~~~~
      |   SAI_HandleTypeDef
In file included from /Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/Wire/src/Wire.h:28,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/DaisyDuino.h:5,
                 from /Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/AudioClass.cpp:2:
/Users/Tony/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/Wire/src/utility/twi.h:101:3: error: 'I2C_HandleTypeDef' does not name a type; did you mean 'SAI_HandleTypeDef'?
  101 |   I2C_HandleTypeDef handle;
      |   ^~~~~~~~~~~~~~~~~
      |   SAI_HandleTypeDef
/Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/AudioClass.cpp: In member function 'DaisyHardware AudioClass::init(DaisyDuinoDevice, DaisyDuinoSampleRate)':
/Users/Tony/Documents/Arduino/libraries/DaisyDuino/src/AudioClass.cpp:53:3: error: 'HAL_Init' was not declared in this scope
   53 |   HAL_Init();
      |   ^~~~~~~~
exit status 1
Error compiling for board Generic STM32H7 Series.
Invalid library found in /Users/Tony/Documents/Arduino/libraries/Firmata: no headers files (.h) found in /Users/Tony/Documents/Arduino/libraries/Firmata

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

stephenhensley, 2021-04-28:
thanks for the additional info. 

I haven't had a chance to get to resolving this, but am hoping to have the time before the end of this week. This is helpful feedback.

In the meantime, I suggest you revert to v1.9 of the STM32Duino support until this is resolved. I've briefly outlined the steps in [this forum post](https://forum.electro-smith.com/t/stm32duino-arduino-core-v2-0-compatibility/1535)

DiegoLiedo, 2021-06-13:
Hi I can confirm this bug with STMDuino v2.0 either on windows 10 with arduino ide 1.8.15 and on Fedora 34 with arduino ide. 
There's also no simple way of downgrading STM32Duino to v1.9 as the boards manager doesn't show versions.
How can I downgrade it?

stephenhensley, 2021-07-12:
This has been fixed with v1.4.0

closed 2021-07-12 by stephenhensley