electro-smith/DaisyDuino #33: DaisyDuino v1.3.0 no audio

DiegoLiedo, 2021-07-03:
Hi, I started playing around with a daisy seed about a month ago.
It took me a while to actually get some sound of it when flashing through arduino IDE.
I did a lot of testing, different IDE versions, different OS, always with the same result: code from the examples compiled and uploaded without problems but sound never existed even though toggling the built in LED did.
After a while I downgraded to DaisyDuino v1.2.0 and everything worked fine. 
Did I miss something about turning sound on on v1.3.0? Or is something broken?

DiegoLiedo, 2021-07-07:
DaisyDuino v.1.4.0 works fine.

stephenhensley, 2021-07-12:
This may have been another issue between the compatibility between the recent update to Stm32duino and DaisyDuino

If everything works on v1.4.0 for you, I'll close this up. Glad you got it working!

closed 2021-07-12 by stephenhensley