electro-smith/DaisyDuino #36: No audio output when using arduino IDE

marcel-licence, 2021-09-30:
After updating the board and the library there is no audio output when uploading a compiled example sketch with arduino.

```
Using board 'GenH7' from platform in folder: C:\Users\...\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0
Using core 'arduino' from platform in folder: C:\Users\...\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0
```

```
Alternatives for DaisyDuino.h: [DaisyDuino@1.4.0]
ResolveLibrary(DaisyDuino.h)
  -> candidates: [DaisyDuino@1.4.0]
```
Compiler and linker is used from this directory:
`\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\xpack-arm-none-eabi-gcc\\10.2.1-1.1/bin/`

The blink example was successful (led was blinking). In addition to that, adding print statements and activation of CDC was also successful. Only the audio seems to be dead.

beserge, 2021-12-06:
Sorry for the delayed response!
I see 'Using board GenH7' in the output there. Are you using the Generic H7 device or do you have the daisy seed selected?
You should also make sure you've got the latest versions of daisyduino and stm32duino.
There's more instructions on all of that [here](https://github.com/electro-smith/DaisyWiki/wiki/1a.-Getting-Started-(Arduino-Edition))

marcel-licence, 2021-12-08:
Thank you for your response. I will check it and come back to you with my results.

DaisyDuino is updated to the last version: 1.4.0. There is no update available in arduino.
"Daisy Seed" can be found as board part number listet in the "STM32 H7" boards. But this was the stm32duino version 2.1 which is now deprecated.

Your manual might be outdated because it refers to the stm32duino version 2.0 which would be outdated: https://github.com/stm32duino/wiki/wiki/Getting-Started

The is a new version be Seeed Studio (not sure if I had compiling issues by trying to use it):
[image]

takumi-ogata, 2023-03-04:
Hi Marcel! 

We're currently going around checking up and was wondering if you were still having this issue. 
DaisyDuino and its documents have been updated since December of 2021 so we recommend giving it a try again. Thanks!

marcel-licence, 2023-03-05:
Hi,

thank you for coming back to me. I've updated DaisyDuino to version 1.6.3 and tried to compile an example.

It looks like that the selected core does not work for me and I have to update or find another:

My configuration:
<A name="STMicroelectronics_stm32_GenH7"></A><B>Core:</B> <A h_ref="https://github.com/stm32duino/Arduino_Core_STM32">STM32 boards groups </A><BR><B>Version:</B> 2.1.0<BR><B>Board:</B> Generic STM32H7 Series<BR><BR><B>Board part number:</B> Daisy Seed<BR><B>Upload method:</B> STM32CubeProgrammer (DFU)<BR><B>U(S)ART support:</B> Enabled (generic 'Serial')<BR><B>USB support (if available):</B> CDC (generic 'Serial' supersede U(S)ART)<BR><B>USB speed (if available):</B> Low/Full Speed<BR><B>Optimize:</B> Smallest (-Os default)<BR><B>Debug symbols:</B> None<BR><B>C Runtime Library:</B> Newlib Nano (default)<BR><BR><B>Used libraries:</B><BR><TABLE><TBODY><TR><TD>Name</TD><TD>Version</TD><TD>Url</TD><TD>Git</TD><TD>Core library</TD></TR><TR><TD>DaisyDuino</TD><TD>1.6.3</TD><TD>https://github.com/electro-smith/DaisyDuino</TD><TD>https://github.com/electro-smith/DaisyDuino.git</TD><TD>False</TD></TR><TR><TD>Wire</TD><TD>1.0</TD><TD>http://www.arduino.cc/en/Reference/Wire</TD><TD></TD><TD>True</TD></TR><TR><TD>Source Wrapper</TD><TD>1.0.1</TD><TD></TD><TD></TD><TD>True</TD></TR></TBODY></TABLE>

The build error which occurred:
```
C:\Users\...\Documents\Arduino\libraries\DaisyDuino\src\AudioClass.cpp:292:13: error: 'PD3' was not declared in this scope; did you mean 'D3'?
  292 |     pinMode(PD3, INPUT_PULLUP);
      |             ^~~
      |             D3
```

marcel-licence, 2023-03-05:
Following configuration worked for me (looks like the Core shouldn't be too old:

<hr>
<a name="STMicroelectronics_stm32_GenH7"></a><b>Core:</b> <a h_ref="https://github.com/stm32duino/Arduino_Core_STM32">STM32 boards groups </a><br />
<b>Version:</b> 2.4.0<br />
<b>Board:</b> Generic STM32H7 Series<br />
<br />
<b>Board part number:</b> Daisy Seed<br />
<b>Upload method:</b> STM32CubeProgrammer (DFU)<br />
<b>U(S)ART support:</b> Enabled (generic 'Serial')<br />
<b>USB support (if available):</b> CDC (generic 'Serial' supersede U(S)ART)<br />
<b>USB speed (if available):</b> Low/Full Speed<br />
<b>Optimize:</b> Smallest (-Os default)<br />
<b>Debug symbols and core logs:</b> None<br />
<b>C Runtime Library:</b> Newlib Nano (default)<br />
<br />
<hr>
I can confirm that the examples now do work without any changes. Thank you!

closed 2023-03-05 by marcel-licence